chipsalliance/fpga-tool-perf #2: Allow "sweeping" of a given VPR parameter

mithro, 2018-07-16:
There is a number of VPR parameters we will want to investigate the best value for soon. I think the best way is to just "sweep" the value and see what it comes up with.

I think this would be similar to the seed stuff you added previously?

One primary argument we are interested in is the `--target_ext_pin_util` argument. This controls how aggressively the cluster will pack things into tiles (which can affect routing).

The help from vpr for this argument;
```

  --target_ext_pin_util TARGET_EXT_PIN_UTIL [TARGET_EXT_PIN_UTIL ...]            
                    Sets the external pin utilization target during clustering.
                    Value Ranges:                                               
                    * 1.0: The packer to pack as densely as possible (i.e. try  
                           to use 100% of cluster external pins)        
                    * 0.0: The packer to pack as loosely as possible (i.e. each
                           block will contain a single mollecule) Values in
                           between trade-off pin usage and packing density
                                                                           
                    Typically packing less densely improves routability, at     
                    the cost of using more clusters. Note that this is only
                    a guideline, the packer will use up to 1.0 utilization if
                    a molecule would not otherwise not in any cluster type.    
                                                                             
                    This option can take multiple specifications in several   
                    formats:
                    * Single Value (e.g. '0.7'): the input pin utilization for 
                                                 all block types (output pin
                                                 utilization defaults to 1.0)   
                    * Double Value (e.g. '0.7,0.8'): the input and output pin
                                                 utilization for all block types
                    * Block Value (e.g. 'clb:0.7', 'clb:0.7,0.8'): the pin
                                                 utilization for a specific
                                                 block type  
                    These can be used in combination. For example:
                       '--target_ext_pin_util 0.9 clb:0.7'                  
                    would set the input pin utilization of clb blocks to 0.7,
                    and all other blocks to 0.9.                         
                     (Default: 1.0)   
```

Another example would be the `--timing_tradeoff` argument (in the placement stage) and the `--criticality_exp` arguments.

```
  --timing_tradeoff TIMING_TRADEOFF
                    Trade-off control between delay and wirelength during   
                    placement. 0.0 focuses completely on wirelength, 1.0
                    completely on timing (Default: 0.5)

  --max_criticality MAX_CRITICALITY
                    Sets the maximum fraction of routing cost derived from delay
                    (vs routability) for any net. 0.0 means no attention is paid
                    to delay, 1.0 means nets on 
```

mcmasterg, 2018-07-20:
@mithro has indicated VPR requests are on hold right now

acomodi, 2020-05-26:
This is now supported. Closing

closed 2020-05-26 by acomodi