chipsalliance/fpga-tool-perf #3: Add support for extracting the VPR "routing information table"

mithro, 2018-07-16:
When VPR is running, it outputs the following table when running placement
```
------- ------- ---------- ---------- ---------- ---------- ------- ---------- -------- ------- ------- ------- ------ --------- ------
      T    Cost Av BB Cost Av TD Cost Av Tot Del P to P Del     CPD       sTNS     sWNS Ac Rate Std Dev R limit    Exp Tot Moves  Alpha
------- ------- ---------- ---------- ---------- ---------- ------- ---------- -------- ------- ------- ------- ------ --------- ------
  0.059  0.9998  1405.0470 44676      1.2679e+05 7.4374     156500000000.000  -1.62e+14 -156500000000.000  0.9980  0.0089 37.0000  1.000     6060  0.500
  0.030  0.9973  1404.2358 45429      1.2656e+05 7.4485     152500000000.000  -1.63e+14 -152500000000.000  0.9941  0.0045 37.0000  1.000    12120  0.500
  0.015  0.9910  1402.7112 44757      1.2695e+05 7.4821     154000000000.000  -1.61e+14 -154000000000.000  0.9875  0.0072 37.0000  1.000    18180  0.500
  0.007  0.9866  1404.5965 45357      1.2669e+05 7.4978     153000000000.000  -1.56e+14 -153000000000.000  0.9752  0.0048 37.0000  1.000    24240  0.500
  0.004  0.9947  1386.8590 46141      1.259e+05  7.4402     150000000000.000  -1.62e+14 -150000000000.000  0.9512  0.0049 37.0000  1.000    30300  0.900
  0.003  0.9878  1385.0031 42649      1.2599e+05 7.4897     160000000000.000  -1.61e+14 -160000000000.000  0.9467  0.0064 37.0000  1.000    36360  0.900
  0.003  0.9941  1384.8312 46297      1.261e+05  7.4572     150000000000.000  -1.62e+14 -150000000000.000  0.9348  0.0050 37.0000  1.000    42420  0.900
  0.003  0.9930  1377.3164 47336      1.2568e+05 7.4401     144000000000.000  -1.67e+14 -144000000000.000  0.9290  0.0080 37.0000  1.000    48480  0.900
  0.002  0.9952  1383.4054 43231      1.2583e+05 7.4167     160500000000.000  -1.65e+14 -160500000000.000  0.9276  0.0050 37.0000  1.000    54540  0.900
  0.002  0.9975  1379.3081 44056      1.2557e+05 7.4138     156000000000.000  -1.62e+14 -156000000000.000  0.9167  0.0052 37.0000  1.000    60600  0.900
  0.002  0.9922  1367.5716 40780      1.2539e+05 7.3734     172500000000.000  -1.64e+14 -172500000000.000  0.9031  0.0046 37.0000  1.000    66660  0.900
  0.002  1.0070  1366.0199 46003      1.2529e+05 7.3377     144500000000.000  -1.55e+14 -144500000000.000  0.8861  0.0060 37.0000  1.000    72720  0.900
  0.002  0.9882  1364.4663 45681      1.2505e+05 7.4373     145500000000.000  -1.66e+14 -145500000000.000  0.8848  0.0051 37.0000  1.000    78780  0.900
  0.001  1.0062  1362.4004 43692      1.2489e+05 7.2775     157000000000.000  -1.62e+14 -157000000000.000  0.8756  0.0045 37.0000  1.000    84840  0.900
  0.001  0.9938  1357.4930 45085      1.2473e+05 7.3188     146500000000.000  -1.57e+14 -146500000000.000  0.8639  0.0046 37.0000  1.000    90900  0.900
  0.001  0.9811  1341.4549 47280      1.244e+05  7.3691     138500000000.000  -1.57e+14 -138500000000.000  0.8408  0.0067 37.0000  1.000    96960  0.900
  0.001  0.9984  1338.5302 45890      1.2418e+05 7.2875     142500000000.000  -1.54e+14 -142500000000.000  0.8282  0.0055 37.0000  1.000   103020  0.900
```

And the following table when running routing;
```
Confirming router algorithm: TIMING_DRIVEN.
---- ------ ------- ------- ------- ------- ----------------- --------------- -------- ---------- ---------- ---------- ---------- --------
Iter   Time    Heap    Heap  Re-Rtd  Re-Rtd Overused RR Nodes      Wirelength      CPD       sTNS       sWNS       hTNS       hWNS Est Succ
      (sec)    push     pop    Nets   Conns                                       (ns)       (ns)       (ns)       (ns)       (ns)     Iter
---- ------ ------- ------- ------- ------- ----------------- --------------- -------- ---------- ---------- ---------- ---------- --------
   1   13.7 1.3e+07 2194666    5088   16982    6450 ( 2.743%)  197170 (56.6%) 117500000000.000 -1.644e+14 -117500000000.000 -3.640e+11 -9000000000.000      N/A
   2   14.9 1.6e+07 2321755    3442   14609    3255 ( 1.384%)  195842 (56.2%) 132000000000.000 -1.687e+14 -132000000000.000 -2.642e+12 -25500000000.000      N/A
   3   16.1 1.9e+07 3057604    2748   12825    3399 ( 1.445%)  198408 (56.9%) 153500000000.000 -1.608e+14 -153500000000.000 -4.625e+12 -25000000000.000      N/A
   4   18.1 2.2e+07 3653020    2691   12794    3529 ( 1.501%)  201321 (57.7%) 150500000000.000 -1.774e+14 -150500000000.000 -1.538e+12 -25000000000.000      N/A
   5   19.2 2.4e+07 4088265    2354   12334    2070 ( 0.880%)  208846 (59.9%) 168500000000.000 -1.903e+14 -168500000000.000 -8.756e+12 -38500000000.000      N/A
   6   20.5 2.8e+07 4885353    2261   12022    1846 ( 0.785%)  209549 (60.1%) 158500000000.000 -2.291e+14 -158500000000.000 -4.228e+12 -25000000000.000      N/A
   7   22.3 3.0e+07 5754771    2242   11875    1524 ( 0.648%)  211368 (60.6%) 153000000000.000 -2.045e+14 -153000000000.000 -2.848e+12 -30000000000.000      N/A
   8   23.9 3.3e+07 7030596    2108   11635    1131 ( 0.481%)  215623 (61.8%) 168500000000.000 -2.440e+14 -168500000000.000 -1.147e+13 -60000000000.000      N/A
   9   21.7 2.8e+07 6330047    2008   11393     763 ( 0.324%)  218738 (62.7%) 253500000000.000 -3.972e+14 -253500000000.000 -1.041e+14 -119500000000.000      N/A
  10   17.9 2.3e+07 5469541    1957   11374     552 ( 0.235%)  218191 (62.6%) 253000000000.000 -2.301e+14 -253000000000.000 -1.306e+14 -119500000000.000       36
  11   18.2 2.3e+07 5481704    1938   11261     452 ( 0.192%)  219443 (62.9%) 256500000000.000 -2.265e+14 -256500000000.000 -1.308e+14 -119500000000.000       30
```
These tables give us an understanding of how well the router / placer is doing. It would be to capture and preserve them.

mithro, 2018-07-18:
FYI These names are;
 * CPD - Critical Path Delay
 * sTNS - Setup Total Negative Slack
 * sWNS - Setup Worst Negative Slack
 * hTNS - Hold Total Negative Slack
 * hWNS - Hold Worst Negative Slack

mcmasterg, 2018-07-20:
@mithro has indicated VPR requests are on hold right now