chipsalliance/fpga-tool-perf #4: Add Murax SoC as test example

mithro, 2018-07-18:
From https://github.com/SpinalHDL/VexRiscv#murax-soc
> Murax is a very light SoC (it fits in an ICE40 FPGA) which can work without any external components.
>
> Depending the CPU configuration, on the ICE40-hx8k FPGA with icestorm for synthesis, the full SoC has the following area/performance :
> * RV32I interlocked stages => 51 Mhz, 2387 LC 0.45 DMIPS/Mhz
> * RV32I bypassed stages => 45 Mhz, 2718 LC 0.65 DMIPS/Mhz

This should be small enough to fit on the `up5k` and the `hx8k` (and the `lp8k`) making a good target which hits all our boards?

We should probably generate the verilog once and push it to a repo so people don't have to use the SpinalHDL stuff.

mcmasterg, 2018-07-20:
@mithro per our chat the other day are you convinced now that existing tests are comprehensive enough? Do you still want this?

mithro, 2018-07-20:
I think we have a okay set of tests. I think that this SoC possibly still fits in a nice space (fitting on the up5k) and we probably want to investigate it anyway?

mithro, 2018-07-20:
Maybe it should be a todo item for myself or @elmsfu so that we know how to add tests to the tool?

PR #6 (2018-07-23): Adding Murax SoC (and misc improvements)
 * Adding Murax SoC (fixes #4).
 * Tested with vpr and nextpnr on hx8k and ct256.
 * Added support for data files (needed for the SoC memory init).
 * Cleanup some whitespace and argument stuff.
 * Small document improvements.
closed 2018-07-23 by mcmasterg