chipsalliance/fpga-tool-perf #8: Support carry / no_carry options?

mithro, 2018-07-30:
It might be interesting to support running `arachne`, `nextpnr` and `vpr` with and without carry support.

It's something which is interesting now but might not be interesting in the future. I'm currently just just modifying the script directly.

Don't know if we could support that for `radiant` or `icecube`?

daveshah1, 2018-07-30:
You could implement this easily for radiant or icecube by using Yosys to presyntheise

JohnDMcMaster, 2018-08-07:
Added support for arachne and nextpnr in 2dc896aeacb12f420917c59de49f83c7d61b2177. I believe VPR doesn't support carry, so it still requires --no-carry (the default for vpr)

Sample usage + output:

```
mcmaster@phantom:~/sf/fpga-tool-perf$ python3 fpgaperf.py --toolchain arachne --project oneblink --device "hx8k" --package "ct256" --carry
Writing to build/ice40-hx8k-ct256_arachne_oneblink_default
Running: yosys -p 'synth_ice40 -top top -blif my.blif' /home/mcmaster/sf/fpga-tool-perf/src/oneblink.v
Running: arachne-pnr -d 8k -P ct256 -o my.asc my.blif
Running: icepack my.asc my.bin
Running: icetime -tmd hx8k my.asc
Design ice40-hx8k-ct256_arachne_oneblink_default
  Family: ice40
  Device: hx8k
  Package: ct256
  Project: oneblink
  Toolchain: arachne
  Strategy: default
  Carry: True
  Seed: default
Timing:
  nop:             0.001
  yosys:           0.431
  arachne-pnr:     1.763
  icepack:         0.176
  bit-all:         2.371
  icetime:         3.937
Max frequency: 176.190 MHz
Resource utilization
  BRAM:                0
  CARRY:               22
  DFF:                 24
  GLB:                 1
  IOB:                 2
  LUT:                 26
  PLL:                 0

mcmaster@phantom:~/sf/fpga-tool-perf$ python3 fpgaperf.py --toolchain arachne --project oneblink --device "hx8k" --package "ct256" --no-carry
Writing to build/ice40-hx8k-ct256_arachne_oneblink_default
Running: yosys -p 'synth_ice40 -top top -nocarry -blif my.blif' /home/mcmaster/sf/fpga-tool-perf/src/oneblink.v
Running: arachne-pnr -d 8k -P ct256 -o my.asc my.blif
Running: icepack my.asc my.bin
Running: icetime -tmd hx8k my.asc
Design ice40-hx8k-ct256_arachne_oneblink_default
  Family: ice40
  Device: hx8k
  Package: ct256
  Project: oneblink
  Toolchain: arachne
  Strategy: default
  Carry: False
  Seed: default
Timing:
  nop:             0.001
  yosys:           0.405
  arachne-pnr:     1.591
  icepack:         0.142
  bit-all:         2.140
  icetime:         3.349
Max frequency: 85.980 MHz
Resource utilization
  BRAM:                0
  CARRY:               0
  DFF:                 24
  GLB:                 1
  IOB:                 2
  LUT:                 31
  PLL:                 0

mcmaster@phantom:~/sf/fpga-tool-perf$ python3 fpgaperf.py --toolchain nextpnr --project oneblink --device "hx8k" --package "ct256" --carry
Writing to build/ice40-hx8k-ct256_nextpnr_oneblink_default
Running: yosys -p 'synth_ice40 -top top ; write_json my.json' /home/mcmaster/sf/fpga-tool-perf/src/oneblink.v
Running: nextpnr-ice40  --hx8k --package ct256 --json my.json --asc my.asc
Running: icepack my.asc my.bin
Running: icetime -tmd hx8k my.asc
Design ice40-hx8k-ct256_nextpnr_oneblink_default
  Family: ice40
  Device: hx8k
  Package: ct256
  Project: oneblink
  Toolchain: nextpnr
  Strategy: default
  Carry: True
  Seed: default
Timing:
  nop:             0.001
  yosys:           0.285
  nextpnr-ice40:   0.304
  icepack:         0.142
  bit-all:         0.732
  icetime:         3.127
Max frequency: 165.550 MHz
Resource utilization
  BRAM:                0
  CARRY:               22
  DFF:                 24
  GLB:                 1
  IOB:                 2
  LUT:                 27
  PLL:                 0

mcmaster@phantom:~/sf/fpga-tool-perf$ python3 fpgaperf.py --toolchain nextpnr --project oneblink --device "hx8k" --package "ct256" --no-carry
Writing to build/ice40-hx8k-ct256_nextpnr_oneblink_default
Running: yosys -p 'synth_ice40 -top top -nocarry ; write_json my.json' /home/mcmaster/sf/fpga-tool-perf/src/oneblink.v
Running: nextpnr-ice40  --hx8k --package ct256 --json my.json --asc my.asc
Running: icepack my.asc my.bin
Running: icetime -tmd hx8k my.asc
Design ice40-hx8k-ct256_nextpnr_oneblink_default
  Family: ice40
  Device: hx8k
  Package: ct256
  Project: oneblink
  Toolchain: nextpnr
  Strategy: default
  Carry: False
  Seed: default
Timing:
  nop:             0.001
  yosys:           0.366
  nextpnr-ice40:   0.450
  icepack:         0.158
  bit-all:         0.975
  icetime:         3.249
Max frequency: 81.880 MHz
Resource utilization
  BRAM:                0
  CARRY:               0
  DFF:                 24
  GLB:                 1
  IOB:                 2
  LUT:                 32
  PLL:                 0
```

closed 2018-08-07 by JohnDMcMaster