chipsalliance/fpga-tool-perf #15: nextpnr doesn't need -nocarry in synth script

daveshah1, 2018-08-05:
nextpnr has stable carry support now, no need to include the `-nocarry` any more.

Also, looks like the VPR synth script runs ABC twice which may give better results than arachne or nextpnr's scripts which only run ABC once. This should probably be using `synth_ice40 -vpr` instead.

mcmasterg, 2018-08-05:
Good to know. Maybe ideally we'd have a global carry flag that would default to true so there is still an option to compare.

re: VPR, I was told there was *possibly* some issue with "synth_ice40 -vpr". I mostly followed what symbilflow-arch-defs was doing and was unclear if the issue had been resolved. I'll try this out soon and see if I run into any snags.

I agree that it would be best to have an apples-to-apples comparison. However, if we think running abc multiple times possibly gives better performance, we should also have an issue to explore this.

mithro, 2018-08-06:
FYI - https://github.com/SymbiFlow/fpga-tool-perf/issues/8

JohnDMcMaster, 2018-08-07:
Fixed 8, default is now to carry

closed 2018-08-07 by JohnDMcMaster