chipsalliance/fpga-tool-perf #17: yosys: investigate multiple abc passes

JohnDMcMaster, 2018-08-07:
I've heard it might be possible to get better abc performance through multiple passes

JohnDMcMaster, 2018-08-07:
TLDR: this took longer and didn't improve QoR. I will not add any options to add extra abc passes

Tried to compare:
```
"synth_ice40 -top %s" % (self.top,),
"write_blif -gates -attr -param my.blif",
```
with
```
"synth_ice40 -top %s" % (self.top,),
"ice40_opt",
"abc",
"opt_clean",
"write_blif -gates -attr -param my.blif",
```

Using arachne toolchain with vexriscv-verilog and 10 different seeds tried on each script. This resulted in the same min and max performance, although the extra abc steps made things take longer:
-One pass: build in 1092.8 to 1395.2 seconds
-Two pass: build in 1174.6 to 1722.3 seconds

closed 2018-08-07 by JohnDMcMaster

daveshah1, 2018-08-07:
Good to know, thanks for testing this anyway!