chipsalliance/fpga-tool-perf #51: Generate nicer reports

kgugala, 2020-02-28:
as suggested by @mithro in https://github.com/SymbiFlow/fpga-tool-perf/pull/50#issuecomment-592180396 we should generate the following reports:

```================================
Settings
================================

+-----------+--------------+
| Setting   | Value        |
+-----------+--------------+
| Family    | xc7          |
| Device    | a35ti        |
| Package   | csg324-1L    |
| Project   | litex-linux  |
| Toolchain | vivado       |
| Strategy  | None         |
| Carry     | False        |
| Seed      | default      |
+-----------+--------------+

================================
FPGA Resource Utilization
================================

+----------+------+--------+---------+
| Resource | Used | Out of | Percent |
+----------+------+--------+---------+
| BRAM     | 40.5 |  8000  |   XXX%  |
| CARRY    | 260  |  8000  |   XXX%  |
| DFF      | 4883 |  8000  |   XXX%  |
| IOB      | 89   |  8000  |   XXX%  |
| LUT      | 6207 |  8000  |   XXX%  |
| PLL      | 1    |  8000  |   XXX%  |
+----------+------+--------+---------+

================================
Clocks
================================

+------------------------------+-----------+---------+-----------+---------+
| Clock Domain                 | Requested | Actual  | Satisfied |  Margin |
+------------------------------+-----------+---------+-----------+---------+
| clock domain soc_pll_clk200  | 200MHz    | 849 MHz | Yes       |  649MHz |
| clock domain soc_pll_clk100  | 100MHz    | 249 MHz | Yes       |  149MHz |
| clock domain soc_pll_clk800  | 800MHz    | 249 MHz | No        | -561MHz |
+------------------------------+-----------+---------+-----------+---------+

================================
Toolchain Resource Usage
================================

+----------------------+-----------+-------------+
|                      | Run Time  | Peak Memory |
| Stage                | (seconds) | (megabytes) |
+----------------------+-----------+-------------+
| preparing            |    0.004  |     100     |
| verilog to bitstream |  178      |    2567     |
|  - synth             |  +XX      |             |
|  - pnr               |  +XX      |             |
|  - bitgen            |  +XX      |             |
+----------------------+-----------+-------------+
```

mithro, 2020-06-25:
@HackerFoo -- what is the status here?

HackerFoo, 2020-06-26:
@mithro It looks like the current reports are very close to this: [example](https://hydra.vtr.tools/build/1426/log)

There are some differences e.g. no "Margin" column in the "Clocks" table, and columns missing from the resource table.

Is there anything missing that should be added?