chipsalliance/fpga-tool-perf #125: Refactor and clean up the build system

acomodi, 2020-05-26:
Currently, there are two things that could be made better, to have a cleaner build system for the tests:

1. Restructure the ways in which the constraints files are selected to run the exhaust script. This would avoid files duplication.
2. Allow picking to have a dependency structure, so that a test is performed using another one's results: this comes in hand when running, for instance, the [fasm2bels flow](https://github.com/SymbiFlow/fpga-tool-perf/pull/124), in which data generated from the `vpr` flow could be re-used, allowing us to have separate fasm2bels tool, which is not incorporated within the VPR flow.

acomodi, 2020-05-26:
To solve 1 the possible way forward is:

- have the projects jsons define which toolchains are supported, and, for each toolchain, a board-related set of constraints file.

E.g.:
```json
{
    "srcs": [
        "src/blinky/blinky.v"
        ],
    "top": "top",
    "name": "blinky",
    "clocks": {
        "clk_i": 10.0
    },
   "toolchains": {
       "vpr": ["src/blinky/constr/arty.pcf", "src/blinky/constr/basys3.pcf"],
       "vpr-fasm2bels": ["src/blinky/constr/arty.pcf", "src/blinky/constr/basys3.pcf"],
       "nextpnr-xilinx": ["src/blinky/constr/arty.xdc", "src/blinky/constr/basys3.xdc"],
       "vivado": ["src/blinky/constr/arty.xdc", "src/blinky/constr/basys3.xdc"]
   }
}
```

This also requires a new file (probably json) that holds boards specific definition, so that information on family, device and package can be queried:

E.g.:
```json
{
    "arty": {
         "family": "xc7",
         "device": "a35t",
         "package": "csg324-1"
    },
    "...": {}
}
```

mithro, 2020-06-25:
@acomodi  - What is the status here?

acomodi, 2020-06-25:
@mithro the first issue is solved.

The lower priority one, but non-trivial with the current test run system, is to be able to get existing build outputs to use for other build runs (e.g. fasm2bels using vpr outputs, so that P&R step do not need to be performed once again).

mithro, 2020-06-25:
Maybe that should be split into a new github issue?