chipsalliance/fpga-tool-perf #133: Build Problems and OS X Support

srtemp, 2020-05-28:
I'm trying to get this toolchain (minus vivado) running on Mac OS 10.14.6, and have run into a few problems; some may be Mac related, and perhaps that's unsupported, but some seem more general.

Following the Readme:
Set up all tools and added them to my path, set appropriate environment variables.
$ source settings.sh
    no such file.  From the git history, it looks like it's been renamed env.sh, so I used that instead.
$ make env
    no such target.  Looks like make conda might be appropriate, since we're setting up a python environment.  So I ran that, and did some manual setup that I don't recall at the moment to get things running.
$ python3 fpgaperf.py --toolchain arachne --project oneblink --device "hx8k" --package "ct256" --board "breakout"
    fpgaperf.py: error: --family argument required.  Looks like it's missing a required flag.  Looking at the next example, I set family to xc7.  This may well be wrong, but it made the tool run.
That ran the Yosys portion successfully, but then generated the following error:
<some yosys output omitted>
End of script. Logfile hash: af0a50ac80, CPU: user 0.38s system 0.06s
Yosys 0.9+1706 (git sha1 2bda51ac, gcc 10.0.1 -fPIC -Os)
Time spent: 44% 12x read_verilog (0 sec), 14% 1x share (0 sec), ...
arachne-pnr -d 8k -P ct256 -q -p empty.pcf oneblink.blif -o oneblink_arachne.asc
oneblink.blif:10: fatal error: unknown directive '.subckt'
make: *** [oneblink_arachne.asc] Error 1
rm oneblink.blif

This looks like something I've run into before when trying to read hierarchical blifs with tools that don't support them.  But at this point I decided to ask for help.
It seems like the readme is out of date; if anyone could direct me on how to properly get started with this tool, I'd really appreciate it.

Thanks,
Scott

acomodi, 2020-05-29:
Thanks for filing this @srtemp. 

Regarding the various README issues you are right, there have been quite some changes lately and the README is out of date, I will update it accordingly. 

Currently, ice40 tests are not being run in CI, therefore it may be that the blif error is a latent bug in fpga-tool-perf. I'll try to add that as well to the benchmarks run in CI.

acomodi, 2020-06-01:
@srtemp FYI, the README has been updated

srtemp, 2020-06-01:
Thanks @acomodi I'll give it another try!
I appreciate you working on this so promptly

HackerFoo, 2020-06-16:
@srtemp If you don't mind using Nix on your Mac, you could try: https://github.com/HackerFoo/nix-symbiflow

Then, in your checked out repo, run:
```
nix-shell /path/to/nix-symbiflow -A fpga-tool-perf --use-prebuilt-symbiflow false
```

It will take a while.

srtemp, 2020-06-16:
@HackerFoo 
Thank you; I'll give that a try.
I just moved to using a VM with 16.04 and have gotten it running, but a native install would be much better.  I appreciate it.