chipsalliance/fpga-tool-perf #165: Automate visualization of results

HackerFoo, 2020-06-18:
Generate a quick overview of the results of CI for each tools and each design
* Example: (it doesn't need to look like this): https://docs.google.com/spreadsheets/d/1QpC63sSstD-53h1ux2F3AhdR2M1DnBZSVzGGfWIOYQw/edit#gid=0
* I suggest using Colab, because it's quick and can be commented on and presented, but feel free to suggest other options.
  * Colab examples: https://drive.google.com/drive/folders/1bYENID_88n9Xw2FQLQORz2IC59xiK3Y9?usp=sharing

Example data: [dataframe.json.zip](https://github.com/SymbiFlow/fpga-tool-perf/files/4801003/dataframe.json.zip)

@acomodi Is there a public link for the dataframe.json files generated by CI?

acomodi, 2020-06-19:
@HackerFoo After each CI run on master/PR an artifact is produced with the dataframe.json. An example is here: https://storage.cloud.google.com/fpga-tool-perf/artifacts/prod/foss-fpga-tools/fpga-tool-perf/continuous/47/20200617-104132/github/fpga-tool-perf/build/dataframe.json

HackerFoo, 2020-06-19:
@acomodi Is there a good way to programmatically get the latest link? I'm guessing it's `https://.../47/${date}-${hash-or-sequential-id}/.../dataframe.json`, but I'm not sure how to get `hash-or-sequential-id`.

acomodi, 2020-06-19:
@HackerFoo I think that is a question for @mithro as I am unsure on how the kokoro instance does upload the artifacts on GCS. If there was the possibility to have a `latest` link, it would be great.

TypingKoala, 2020-06-19:
Are the columns in the example sheet all the important metrics? Do the metrics you want to display vary between runs?

HackerFoo, 2020-06-19:
@TypingKoala those are a good start, but the design should be flexible enough to add more metrics if needed. Once we have a good overview of the most important metrics, we can create issues to track improvements (such as tracking trends over time.)

TypingKoala, 2020-06-22:
@HackerFoo I've created a proof-of-concept here that ingests from Hydra or GCS, cleans, normalizes, and outputs the information as a CSV that can be exported directly to Google Sheets. Is this along the lines of what you are looking for? Is there any color coding that would be helpful in the output? 

https://colab.research.google.com/drive/1NXLYCYtY4YOslg0lbGryRfGdjKIeYZ1V?usp=sharing

https://docs.google.com/spreadsheets/d/1woQR-Ag2jQCDWpQE5ko89d5jdvkZswYSxb4DmHlBE5o/edit?usp=sharing

HackerFoo, 2020-06-22:
@TypingKoala Could you embed a table into the Colab instead of generating CSV? Like [this](https://colab.research.google.com/drive/11Cf5Mhaa4eu7KygkuOorxe_9OxlS7VqG#scrollTo=H69DZdvr2I8J).

Color coding might be nice, with the vendor toolchain (e.g. Vivado) as the baseline, red for worse, green for better.

TypingKoala, 2020-06-24:
@HackerFoo I've implemented those changes. Please take a look at the [colab](https://colab.research.google.com/drive/1NXLYCYtY4YOslg0lbGryRfGdjKIeYZ1V?usp=sharing#scrollTo=vRAsvLDAKqoO) now. I've added an embedded table with styling to reflect the relative performance compared to the Vivado toolchain for every project.

TypingKoala, 2020-08-04:
Development is in progress in a [separate repo](https://github.com/SymbiFlow/FPGA-Tool-Performance-Visualization-Library). Once [issue #10](https://github.com/SymbiFlow/FPGA-Tool-Performance-Visualization-Library/issues/10) is resolved in that repo, this issue can be closed.

closed 2021-01-05 by HackerFoo