chipsalliance/fpga-tool-perf #168: Test Failure: murax-vivado-yosys-basys3

HackerFoo, 2020-06-23:
https://hydra.vtr.tools/jobset/dusty/fpga-tool-perf#tabs-jobs

```
ERROR: [DRC INBB-3] Black Box Instances: Cell 'murax/system_uartCtrl/uartCtrl_1/rx/bufferCC_3/$auto$simplemap.cc:467:simplemap_dffsr$25243' of type 'murax/system_uartCtrl/uartCtrl_1/rx/bufferCC_3/$auto$simplemap.cc:467:simplemap_dffsr$25244/$_DFFSR_PPP_' has undefined contents and is considered a black box.  The contents of this cell must be defined for opt_design to complete successfully.
ERROR: [DRC INBB-3] Black Box Instances: Cell 'murax/system_uartCtrl/uartCtrl_1/rx/bufferCC_3/$auto$simplemap.cc:467:simplemap_dffsr$25244' of type 'murax/system_uartCtrl/uartCtrl_1/rx/bufferCC_3/$auto$simplemap.cc:467:simplemap_dffsr$25244/$_DFFSR_PPP_' has undefined contents and is considered a black box.  The contents of this cell must be defined for opt_design to complete successfully.
```

acomodi, 2020-06-23:
This is actually working on kokoro CI

acomodi, 2020-06-25:
@HackerFoo What version of yosys is being used in for this test? I think the issue is that, if the latest yosys is used, different yosys flags need to be applied during synthesis.

HackerFoo, 2020-06-26:
@acomodi https://github.com/YosysHQ/yosys/commit/8f1a32064639fa17d67bda508df941c8846a0664

[Updating edalize](https://hydra.vtr.tools/api/scmdiff?branch=&rev2=d2de2b74e9c3044ce8b03445c562cf7a065b35fe&uri=https%3A%2F%2Fgithub.com%2FHackerFoo%2Fnix-symbiflow.git&rev1=7101900484dc905e2ad31fec3ad12da9a5b1af92&type=git) fixed this.

closed 2020-06-26 by HackerFoo