chipsalliance/fpga-tool-perf #179: Problems with Miniconda when Previously Installed

ryancj14, 2020-07-10:
When one runs `make conda` within the fpga-tool-perf repository, miniconda is installed in the `fpga-tool-perf/env` directory. Then, an environment called `fpga-tool-perf-env` is created and filled with the needed packages. The issue is that miniconda may already be installed on the computer.

If Miniconda has previously been installed on the machine, then running `make conda` succeeds in creating the `fpga-tool-perf-env` environment. However, this environment is attached to the original miniconda and stripped of its name, so that the only way to activate it (in my case) is by running `conda activate /home/student/env/conda/envs/fpga-tool-perf-env` as shown below:
[image]
Weirdly activating with the name does work, if you activate the original base first (which has also been stripped of its name).
[image]

There are two primary issues with this:
1. The user's previous miniconda environments will be stripped of their names and harder to access.
2. fpga-tool-perf's `env.sh`, which is used to activate the conda environment, will no longer work.

I think that a possible solution to this issue could be to have it check in the makefile whether or not conda is already installed, and if it is, to not redownload it, and just add the new environment to it. 

This should make `env.sh` work again as well with just one small change on line 5, to have it simply run `conda activate fpga-tool-perf-env` instead of referencing the conda activate command within the fpga-tool-perf files.

acomodi, 2020-07-10:
@ryancj14 Thanks for spotting this. As a work yet to be done, we would soon move to use the `conda-make-env` repository to have conda managed in a more robust way (See issue https://github.com/SymbiFlow/fpga-tool-perf/issues/163).

I have a local work in progress that does that.

mithro, 2020-07-10:
See also https://github.com/SymbiFlow/make-env/issues/8

closed 2021-04-27 by ryancj14