chipsalliance/fpga-tool-perf #189: Adding wiki to fpga-tool-perf

ryancj14, 2020-07-27:
The wiki that I have created is located at: https://github.com/ryancj14/fpga-tool-perf/wiki 

However, I don't have access to the wiki here. Is there a way to pull request to the wiki. I believe only owners with direct editing access can edit the wiki directly. Others could easily clone the files from the example wiki, and upload them here if they have access. Let me know how to proceed with this.

mithro, 2020-07-28:
I made it so that anyone can modify the wiki at https://github.com/SymbiFlow/fpga-tool-perf/wiki

mithro, 2020-07-28:
Please add your contents to our wiki.

Thanks for your efforts!

ryancj14, 2020-07-30:
Ok I will be able to do by Saturday. Thank you!

ryancj14, 2020-08-03:
I've successfully added the wiki.

closed 2020-08-03 by ryancj14