chipsalliance/fpga-tool-perf #240: Use the Antmicro BES library to publish fpga-tool-perf results into https://source.cloud.google.com/results

mithro, 2020-10-05:
https://github.com/antmicro/distant-bes

mithro, 2020-10-06:
FYI -- @pgielda

pgielda, 2020-10-06:
Is it something that is already running somewhere, e.g. on Travis and we just have to push the results? Because it seems that the Travis job is really simple and is not really running fpga-tool-perf?
Or do you mean a completely new CI?

acomodi, 2020-10-06:
@pgielda @mithro Currently, fpga-tool-perf tests are running on Kokoro. There are two types of builds, one is run when submitting PRs (presubmit) and the other is run when merging on master (continuous).

The continuous CI is performing a wider range of tests (e.g. run multiple times with different seeds, etc.).
I believe that we can add the BES library usage after the completion of the tests on the continuous CI

pgielda, 2020-10-06:
@acomodi can you send me some links? (Priv is fine)

mithro, 2020-10-06:
https://source.cloud.google.com/results/invocations/6cac6cfd-6df7-4eac-9d52-3ab1ecc4941f/targets

mithro, 2020-10-06:
If you look in the ["log"](https://source.cloud.google.com/results/invocations/6cac6cfd-6df7-4eac-9d52-3ab1ecc4941f/log) page you get the following table;
```
+--------------------+--------------------------+---------+---------------+--------------+-------------+----------+---------------------+--------+
| Project            | Toolchain                | Family  | Part          | Board        | Build Type  | Build N. | Options             |        |
+--------------------+--------------------------+---------+---------------+--------------+-------------+----------+---------------------+--------+
| baselitex          | nextpnr-xilinx-fasm2bels | xc7     | a35tcsg324-1  | arty         | generic-all | 000      | xdc_carry-n         | failed |
| baselitex          | nextpnr-xilinx           | xc7     | a35tcsg324-1  | arty         | generic-all | 000      | xdc_carry-n         | passed |
| baselitex          | vivado                   | xc7     | a35tcsg324-1  | arty         | generic-all | 000      | xdc_carry-n         | passed |
| baselitex          | vpr-fasm2bels            | xc7     | a35tcsg324-1  | arty         | generic-all | 000      | xdc_carry-n         | passed |
| baselitex          | vpr                      | xc7     | a35tcsg324-1  | arty         | generic-all | 000      | sdc_xdc_carry-n     | passed |
| baselitex          | yosys-vivado             | xc7     | a35tcsg324-1  | arty         | generic-all | 000      | xdc_carry-n         | passed |
| blinky             | nextpnr-xilinx-fasm2bels | xc7     | a35tcpg236-1  | basys3       | generic-all | 000      | sdc_xdc_carry-n     | passed |
| blinky             | nextpnr-xilinx-fasm2bels | xc7     | a35tcsg324-1  | arty         | generic-all | 000      | sdc_xdc_carry-n     | passed |
| blinky             | nextpnr-xilinx           | xc7     | a35tcpg236-1  | basys3       | generic-all | 000      | sdc_xdc_carry-n     | passed |
| blinky             | nextpnr-xilinx           | xc7     | a35tcsg324-1  | arty         | generic-all | 000      | sdc_xdc_carry-n     | passed |
| blinky             | vivado                   | xc7     | a35tcpg236-1  | basys3       | generic-all | 000      | sdc_xdc_carry-n     | passed |
| blinky             | vivado                   | xc7     | a35tcsg324-1  | arty         | generic-all | 000      | sdc_xdc_carry-n     | passed |
| blinky             | vpr-fasm2bels            | xc7     | a35tcpg236-1  | basys3       | generic-all | 000      | sdc_xdc_carry-n     | passed |
| blinky             | vpr-fasm2bels            | xc7     | a35tcsg324-1  | arty         | generic-all | 000      | sdc_xdc_carry-n     | passed |
| blinky             | vpr                      | xc7     | a35tcpg236-1  | basys3       | generic-all | 000      | sdc_xdc_carry-n     | passed |
| blinky             | vpr                      | xc7     | a35tcsg324-1  | arty         | generic-all | 000      | sdc_xdc_carry-n     | passed |
| blinky             | yosys-vivado             | xc7     | a35tcpg236-1  | basys3       | generic-all | 000      | sdc_xdc_carry-n     | passed |
| blinky             | yosys-vivado             | xc7     | a35tcsg324-1  | arty         | generic-all | 000      | sdc_xdc_carry-n     | passed |
| bram-n1            | nextpnr-xilinx-fasm2bels | xc7     | a35tcpg236-1  | basys3       | generic-all | 000      | sdc_xdc_carry-n     | failed |
| bram-n1            | nextpnr-xilinx           | xc7     | a35tcpg236-1  | basys3       | generic-all | 000      | sdc_xdc_carry-n     | failed |
| bram-n1            | vivado                   | xc7     | a35tcpg236-1  | basys3       | generic-all | 000      | sdc_xdc_carry-n     | passed |
| bram-n1            | vpr-fasm2bels            | xc7     | a35tcpg236-1  | basys3       | generic-all | 000      | sdc_xdc_carry-n     | passed |
| bram-n1            | vpr                      | xc7     | a35tcpg236-1  | basys3       | generic-all | 000      | sdc_xdc_carry-n     | passed |
| bram-n1            | yosys-vivado             | xc7     | a35tcpg236-1  | basys3       | generic-all | 000      | sdc_xdc_carry-n     | passed |
| bram-n2            | nextpnr-xilinx-fasm2bels | xc7     | a35tcpg236-1  | basys3       | generic-all | 000      | sdc_xdc_carry-n     | failed |
| bram-n2            | nextpnr-xilinx           | xc7     | a35tcpg236-1  | basys3       | generic-all | 000      | sdc_xdc_carry-n     | failed |
| bram-n2            | vivado                   | xc7     | a35tcpg236-1  | basys3       | generic-all | 000      | sdc_xdc_carry-n     | passed |
| bram-n2            | vpr-fasm2bels            | xc7     | a35tcpg236-1  | basys3       | generic-all | 000      | sdc_xdc_carry-n     | passed |
| bram-n2            | vpr                      | xc7     | a35tcpg236-1  | basys3       | generic-all | 000      | sdc_xdc_carry-n     | passed |
| bram-n2            | yosys-vivado             | xc7     | a35tcpg236-1  | basys3       | generic-all | 000      | sdc_xdc_carry-n     | passed |
| bram-n3            | nextpnr-xilinx-fasm2bels | xc7     | a35tcpg236-1  | basys3       | generic-all | 000      | sdc_xdc_carry-n     | failed |
| bram-n3            | nextpnr-xilinx           | xc7     | a35tcpg236-1  | basys3       | generic-all | 000      | sdc_xdc_carry-n     | failed |
| bram-n3            | vivado                   | xc7     | a35tcpg236-1  | basys3       | generic-all | 000      | sdc_xdc_carry-n     | passed |
[... cut ...]
| vexriscv           | yosys-vivado             | xc7     | z010clg400-1  | zybo         | generic-all | 000      | xdc_carry-n         | passed |
+--------------------+--------------------------+---------+---------------+--------------+-------------+----------+---------------------+--------+
| Passed:            | 143                      | Failed: | 44            |              |             |          |                     | 76%    |
+--------------------+--------------------------+---------+---------------+--------------+-------------+----------+---------------------+--------+
```

pgielda, 2020-10-06:
Ok, makes sense.