chipsalliance/fpga-tool-perf #262: Enable VPR toolchain for baselitex-nexys-video

wtatarski, 2020-11-03:
Below error needs to be fixed for successful launch of baselitex-nexys-video with VPR:
```
  File "/fpga-tool-perf/env/conda/envs/symbiflow-env/lib/python3.8/site-packages/prjxray/fasm_assembler.py", line 190, in parse_fasm_filename
    raise FasmLookupError('\n'.join(missing_features))
prjxray.fasm_assembler.FasmLookupError: Segment DB RIOI3, key RIOI3.IOI_OCLKM_0.IOI_IMUX31_1 not found from line 'RIOI3_X105Y145.IOI_OCLKM_0.IOI_IMUX31_1'
Makefile:43: recipe for target 'top.bit' failed
```

litghost, 2020-11-03:
It is a little suspicious that the clock for that RIOI3 is coming from the interconnect, though I do agree the bits are missing.  We should likely investigate adding the bits and figuring out why the router is routing the clock via the interconnect.  

Is this a case where an inverter needs to be absorbed into the OLOGIC tile, but hasn't been?  So the routing looks like `CLK` -> `LUT1/INV` -> `OLOGIC`, instead of `CLK` -> `OLOGIC`?

acomodi, 2020-11-03:
> Is this a case where an inverter needs to be absorbed into the OLOGIC tile

Looking at the design, there is no OSERDES that has an inverted clock, but we need to further investigate this issue.

mithro, 2020-11-03:
What does https://github.com/SymbiFlow/fpga-tool-perf/pull/257 do?

acomodi, 2020-11-04:
@mithro it adds the test design, for now built with vivado and yosys-vivado, until we find a solution to this issue.

HackerFoo, 2020-11-12:
Could I get a statement whether we can currently generate a working bitstream using SymbiFlow & VPR for `baselitex` on `nexys-video`? If so, is there a working test somewhere? Thanks.

wtatarski, 2020-11-12:
@HackerFoo A this moment generating bitstream using VPR is not possible due to issue which I mentioned in this issue.