chipsalliance/fpga-tool-perf #271: Daisho USB design failing

acomodi, 2020-11-17:
With the latest database from conda, the daisho-usb3 design fails in symbiflow with the following error:

```
prjxray.fasm_assembler.FasmInconsistentBits: FASM line "CMT_TOP_R_UPPER_T_X8Y200.PLL_CLK_FREQ_BB1_NS_ACTIVE" wanted to set bit (156, 74, 1) but was cleared by FASM line "CMT_TOP_R_UPPER_T_X8Y200.CMT_TOP_R_UPPER_T_PLLE2_CLKIN2.CMT_TOP_R_UPPER_T_PLLE2_CLK_IN2_INT"
Makefile:43: recipe for target 'top.bit' failed
```

litghost, 2020-11-17:
This is likely a bug in the PLL pip fuzzer.  The MMCM pip fuzzer does a better job of decoupling the bits.  @mkurc-ant is this something you can help with?

mkurc-ant, 2020-11-17:
I can take a look at that. Besides that some PLL PIPs are missing in the issue: https://github.com/SymbiFlow/symbiflow-arch-defs/issues/1750. So anyways PLL PIP fuzzer is worth revisiting.

mkurc-ant, 2020-11-24:
@acomodi This is weird. These two FASM features do not even refer to the same segbits. Can you point me out to the design?

acomodi, 2020-11-24:
@mkurc-ant [This](https://github.com/SymbiFlow/fpga-tool-perf/blob/master/project/daisho-usb3.json) is the top level.

You can run that with the following, prior to having set up the toolperf environment:

```
./fpgaperf.py --toolchain vpr --board nexys-video --project daisho-usb3
```

mkurc-ant, 2020-11-24:
Ok I've found the issue. These features conflict for `segbits_cmt_top_r_upper_t` but not for `segbits_cmt_top_l_upper_t`.