chipsalliance/fpga-tool-perf #295: Yosys-vivado failure on picosoc-spimemio

acomodi, 2020-12-09:
There is a yosys-vivado toolchain failure in one of the test projects, which is causing CI to fail. This is most probably due to the latest yosys package, which has been updated to use upstream yosys:

```
ERROR: picosoc-spimemio yosys-vivado basys3 test has failed (build type generic, build nr. 000)

ERROR MESSAGE:  [...]
7d04310e132\$lut for cells of type $lut.
Using template $paramod$5678aabaf38638ff6c60e03d865caa8704f4fb95\$lut for cells of type $lut.
Using template $paramod$500825ce3d3051f95f0970672a43b19de1ca0012\$lut for cells of type $lut.
Using template $paramod$3752cc190b504e0d74d40c2d0e5e82c39cb5ff29\$lut for cells of type $lut.
Using template $paramod$21efccf29b21389022231a5ef010f482e0b353f1\$lut for cells of type $lut.
No more expansions possible.

3.48. Executing XILINX_DFFOPT pass (optimize FF control signal usage).
terminate called after throwing an instance of 'std::out_of_range'
  what():  dict::at()
Aborted (core dumped)
picosoc-spimemio.mk:13: recipe for target 'picosoc-spimemio.edif' failed

```

daveshah1, 2020-12-09:
If this is happening with unmodified upstream Yosys, please report it as a bug there - this definitely shouldn't be failing this way even if hypothetically there is a problem with the design/flags.

acomodi, 2020-12-09:
@daveshah1 Sure, I wanted to keep track of this here, and,  before opening an issue upstream I wanted to debug what might be the cause of the failure (maybe is a flags/design issue indeed).