chipsalliance/fpga-tool-perf #328: CI is broken in a number of ways?

mithro, 2021-07-02:
https://source.cloud.google.com/results/invocations/88d65c97-05ba-4d83-9a8d-79fbd40f31c1/targets/foss-fpga-tools%2Ffpga-tool-perf%2Fcontinuous/log

mithro, 2021-07-07:
@acomodi Do you know if FPGA Perf tool has "allowed failures" and making sure things don't regress?

acomodi, 2021-07-07:
@mithro CI works in a way that all the possible toolchain/devices/designs are tested. It is expected that some of them won't build successfully, and there are combinations which *have* to pass, and if they fail, the build fails as well.
Take the following for instance: https://github.com/SymbiFlow/fpga-tool-perf/blob/9f5850793ad98f98101792fa3564e894bf60aa1f/project/baselitex.json#L19

If any of the required toolchain fail, the build will be considered failed as well.

mithro, 2021-07-07:
Okay great, we definitely want to make sure that things we care about are not regressing.

mithro, 2021-07-07:
Can we get links posted in the CI and the dashboards linked?