chipsalliance/fpga-tool-perf #402: Results visualization: add possibility to hide lines

acomodi, 2022-04-13:
At the moment there is no possibility to hide the lines in the charts corresponding to the various toolchains.

[image]

The idea is to be able to click on the colored line belonging to a particular toolchain and disable all the lines corresponding to the particular toolchain.

For context, the [results](https://github.com/chipsalliance/fpga-tool-perf/tree/master/results) directory contains the code to gather and generate the HTML representation of all the results produced by the FPGA tool perf CI.

SiyaoIsHiding, 2023-03-09:
I would love to contribute to this issue :D. May u assign this issue to me?

SiyaoIsHiding, 2023-03-17:
@acomodi Just in case u didn't receive the notification

kgugala, 2023-03-18:
Hi @SiyaoIsHiding I've assigned the issue to you

SiyaoIsHiding, 2023-03-19:
@kgugala & @acomodi, 
Hello, I am working on this issue.

May I ask what should I put into the environment variable `RESULTS_DIR` when I build the visualization?
The only clue I found is the CI file
```yml
      - name: Use gcloud CLI
        run: |
          cd results
          mkdir build-results
          gsutil cp gs://${{ secrets.GCP_STORAGE_BUCKET }}/*.json.gz build-results
```
But I do not have the access to your gcloud so I still don't know what I should put into this `build-results` folder.

Besides, may you simply give me an example of this `build-results` folder so that I can test the JS codes? I probably only need to modify the JS script inside the result folder, so I do not need to reproduce all your environment.

Thank you for your time and any advice will be appreciated.

Best,
Jane

tmichalak, 2023-03-20:
Hi @SiyaoIsHiding, thanks for looking into this issue. Let me take a look and get back to you.

tmichalak, 2023-03-20:
@SiyaoIsHiding so actually for the purpose of solving this issue you could simply generate the data source files using the [exhaust script](https://github.com/chipsalliance/fpga-tool-perf/blob/main/README.md#exhaustive-build), copy the resulting *.json.gz to the created `build-results` directory and run `make preprocess` and `make build` in the [results](https://github.com/chipsalliance/fpga-tool-perf/tree/main/results) directory.
However, for simplicity I am uploading two *.json.gz files with the latest results. Copy them to the `build-results` directory and run the `process` and `build` targets. 
[results-20230320-165233.json.gz](https://github.com/chipsalliance/fpga-tool-perf/files/11023158/results-20230320-165233.json.gz)
[results-20230320-085242.json.gz](https://github.com/chipsalliance/fpga-tool-perf/files/11023159/results-20230320-085242.json.gz)

SiyaoIsHiding, 2023-03-20:
Thank you very much. I will work on it immediately.

palaniappan-r, 2024-02-22:
Hi, if the issue is still open, can I pick this up?

SiyaoIsHiding, 2024-02-22:
yeah sure pls