chipsalliance/fpga-tool-perf #516: Add support for RapidWright

tmichalak, 2022-11-10:
The target is to be able to measure the performance of various routing algorithms done with the help of RapidWright.
However, first we need to integrate RW with the fpga-tool-perf.

tmichalak, 2022-11-10:
@clavin-xlnx could you please update the description and add more details and describe which algorithms you would like to add?

clavin-xlnx, 2022-11-10:
RapidWright has a router called `RWRoute`.  It only supports UltraScale+ devices and can be run in both timing-driven and non-timing-driven modes and depends on the design being fully placed (and intra-site nets are routed).  From what I remember, `nextpnr-interchange` does not yet have the ability to output a placed (but not routed) design yet.  So, to run `RWRoute`, the design would need to be unrouted first.  For example, if a from `nextpnr-interchange` was placed and routed and saved as an Interchange fileset, `RWRoute` could be run with the following RapidWright code: 

```java
        EDIFNetlist netlist = LogNetlistReader.readLogNetlist(logNetlistFileName);
        Design design = PhysNetlistReader.readPhysNetlist(physNetlistFileName, netlist);

        // TODO - Remove once nextpnr can export placed-only design
        design.unrouteDesign();
        
        RWRoute.routeDesignFullNonTimingDriven(design);
        // Timing-driven option: RWRoute.routeDesignFullTimingDriven(design);
```